wilds/hass-atonstorage #1: Wrong serial numer of ath-6k-tl inverter

pierpaolo1972, 2022-11-22:
hi I have an Aton ATH-6K-TL inverter, I have installed the integration. When I enter the serial number it gives me an error. the S/N is something like 014F with other numbers. That's 15 characters in total. where am i wrong?

wilds, 2022-11-24:
Hi @pierpaolo1972 
can you try the new version? I think yout error is related to #2

pierpaolo1972, 2022-11-24:
Thanks for the update.
I uploaded version 1.0.2 but when I install the integration it gives me an error.
 "Impossibile caricare il flusso di configurazione: Failed dependencies httpx" 
I've tried uninstalling everything and reinstalling but it doesn't work. I don't know if it's my problem.

wilds, 2022-11-24:
Sorry. I rebuild the version. Now it should work

pierpaolo1972, 2022-11-24:
Ok, now it no longer gives the "Failed dependencies httpx" error. Load it form but if I put my data it doesn't go on. "[%key:common::config_flow::error::unknown%]"
Name and password are those used to access the site. the serial number are the 12 characters that I find on the aton page and on the inverter. but it doesn't work

wilds, 2022-11-24:
I think it's fixed now

pierpaolo1972, 2022-11-24:
It's working. gooood job. Grande, complimenti !! thank you so much for your time and work!!!

wilds, 2022-11-24:
Grazie!

closed 2022-11-24 by wilds