wilds/hass-atonstorage #3: Failed dependencies httpx

marcoCasamento, 2022-11-24:
Hi, 
I've just tried the latest version, but unfortunately it throw an error trying to add the integration:

> Impossibile caricare il flusso di configurazione: Failed dependencies httpx

Following message came from te log
Logger: homeassistant.setup
Source: setup.py:168
First occurred: 11:00:48 (1 occurrences)
Last logged: 11:00:48

Setup failed for httpx: Integration not found.

marcoCasamento, 2022-11-24:
Thanks for your integration, your time, your assistance.
We're making progress, now it complains about the serialNumber field of Controller.

Questo errore ha avuto origine da un'integrazione personalizzata.

Logger: custom_components.atonstorage.config_flow
Source: custom_components/atonstorage/config_flow.py:62
Integration: AtonStorage (documentation, issues)
First occurred: 11:50:29 (1 occurrences)
Last logged: 11:50:29

Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/atonstorage/config_flow.py", line 62, in async_step_user
    await self.async_set_unique_id(slugify(controller.serialNumber))
AttributeError: 'Controller' object has no attribute 'serialNumber'

wilds, 2022-11-24:
fixed

closed 2022-11-24 by wilds