wilds/hass-atonstorage #4: REST result could not be parsed as JSON

marcoCasamento, 2022-11-24:
It seems that some calls returns something that couldn't be parsed as a JSON. 
I'll try the integration in a dev environment ASAP, but in the meanwhile I'm wondering if anyone else is experiencing the same behavior.

Questo errore ha avuto origine da un'integrazione personalizzata.

Logger: custom_components.atonstorage.controller
Source: custom_components/atonstorage/controller.py:114
Integration: AtonStorage (documentation, issues)
First occurred: 13:17:35 (144 occurrences)
Last logged: 14:29:46

REST result could not be parsed as JSON

wilds, 2022-11-24:
I don't have your issue and I can't reproduce. I'm waiting for the log of the error

marcoCasamento, 2022-11-24:
I'm sorry I've thought I could have been able to reproduce the issue in VSCode Dev Container, but something is wrong with my configuration as HomeAssistant complains that Python 3.10 is not installed and import from __future__ didn't help.

I'm actullay monitoring two different inverter with two different username/pwd.
Giving that you said you were unable to reproduce the issue, I've tried to remove an instance of the configuration and it works.
I believe that the problem arise if you configure more than one inverter.

How can I provide you some useful logs? changing loglevel maybe ? 

I'm sorry I'm kind of a newbie on homeassistant and python

wilds, 2022-11-24:
> I'm actullay monitoring two different inverter with two different username/pwd.
this is the problem, at the moment the integration not support multiple inverters. I plan to add this feature ASAP

marcoCasamento, 2022-11-25:
Thanks for your effort. The integration is already a big step forward in gathering useful data from the inverter.

Lantoit, 2023-08-02:
Same problem:

This error originated from a custom integration.

Logger: custom_components.atonstorage.controller
Source: custom_components/atonstorage/controller.py:114
Integration: AtonStorage (documentation, issues)
First occurred: 12:24:36 (2 occurrences)
Last logged: 12:26:34

REST result could not be parsed as JSON

Then the integration disappear from HA

Lantoit, 2023-08-02:
[image]

closed 2023-08-06 by wilds