wilds/hass-atonstorage #7: info

pierpaolo1972, 2023-01-26:
Ciao, la lettura dei dati dell'inverter da home assistant, può in qualche modo interferire con il funzionamento dell'inverter stesso?
Hi, can reading the inverter data from the home assistant interfere with the operation of it?

maatx, 2023-01-26:
I dati vengono presi dal cloud, sarebbe fantastico se si potessero prendere direttamente dall'inverter bypassando il cloud. Quello che ho notato io dall'ultimo aggiornamento del firmware  sul mio impianto è un comportamento non coerente della batteria, ma ho già chiesto di far controllare come mai ha questo strano comportamento.

The data is taken from the cloud, it would be great if it could be taken directly from the inverter bypassing the cloud. What I've noticed since the last firmware update on my system is an inconsistent behavior of the battery, but I've already asked to have it checked why it has this strange behavior.

pierpaolo1972, 2023-01-26:
Anche io ho notato la non corrispondenza della percentuale. l'ho segnalato all'assistenza (non Aton) ma non hanno ancora saputo darmi una risposta e non volevo che venisse fuori che il mal funzionamento derivasse in qualche modo da home assistant.

maatx, 2023-01-26:
Ti assicuro che non viene da Home Assistant perché il problema si evidenzia anche nella loro web app sul sito e anche sulla app per smartphone. Io ho contattato sia il mio installatore (l'impianto ha pochi giorni di vita) sia direttamente la Aton. Inviandogli anche i grafici che dimostrano cosa succede. Li metto anche qui
[image]
[image]

pierpaolo1972, 2023-01-26:
Ti ringrazio. Io da qualche settimana non vado sopra il 95% e l scarica è molto irregolare. Ma se mi confermi che anche tu hai notato questa cosa è un "buon segno". Non sapevo dell'aggiornamento del firmware e l'assistenza sta brancolando nel buio dicendo tutto e il contrario di tutto.

maatx, 2023-01-26:
Ah, me l'hanno _venduta_ come "aggiornamento firmware" poi non so se si è trattato solo di una sistemazione della configurazione, prima non vedevo alcuni dati come ad esempio il numero di batterie. Il sistema sembra abbastanza chiuso e per ora non voglio mettermi a fare casini :) Se fosse regolare mi andrebbe bene anche così, ci sono abbastanza dati per monitorarlo anche con HA. I dati più essenziali hanno un timing di aggiornamento accettabile di qualche decina di secondi. La cosa più importante è la carica batteria ed è proprio qui che ci sono problemi, non si sa mai quale è reale visto i salti che fa. Ho solo notato che non scarica oltre il 20% probabilmente per preservarne la funzionalità e mi sta bene solo che se poi salta improvvisamente a oltre il 90% e non si capisce cosa succede. Uso anche l'altra integrazione e i dati sono ovviamente uguali.

pierpaolo1972, 2023-01-26:
Per il 20% ti confermo che è il valore minimo. Chiesi ad Aton alcune informazioni e mi risposero :"Nella mobile App tuttavia è presente l’icona di riempimento, consideri che la batteria ha una profondità di scarica del 80% in quanto va da uno stato di carica minimo del 20% ad un massimo del 100%, quindi ogni barretta corrisponde circa al 9%."

maatx, 2023-01-26:
a me è andata anche sotto il 20% fino almeno al 15 o 12

closed 2023-01-26 by pierpaolo1972