wilds/hass-atonstorage #12: Warning in all sensors "max_power"

maatx, 2023-08-05:
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:505
Integration: Sensore (documentation, issues)
First occurred: 10:12:44 (4 occurrences)
Last logged: 10:12:44

Entity sensor.max_power_sold (<class 'custom_components.atonstorage.sensor.AtonStorageSensorEntity'>) is using state class 'total_increasing' which is impossible considering device class ('power') it is using; expected None or one of 'measurement'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.panels_max_power (<class 'custom_components.atonstorage.sensor.AtonStorageSensorEntity'>) is using state class 'total_increasing' which is impossible considering device class ('power') it is using; expected None or one of 'measurement'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.battery_max_power (<class 'custom_components.atonstorage.sensor.AtonStorageSensorEntity'>) is using state class 'total_increasing' which is impossible considering device class ('power') it is using; expected None or one of 'measurement'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.max_power_bought (<class 'custom_components.atonstorage.sensor.AtonStorageSensorEntity'>) is using state class 'total_increasing' which is impossible considering device class ('power') it is using; expected None or one of 'measurement'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

closed 2023-08-17 by wilds

MrZax75, 2023-09-04:
sensor.panels_max_power riporta un valore totalmente anomalo (ho 6.4 kwp di pannelli e parte da oltre 32kw)

wilds, 2023-09-05:
è un valore che arriva direttamente dai server aton. Credo sia un valore cumulato relativa all'intera giornata

MrZax75, 2023-09-05:
Ciao. Grazie per la risposta. Ma fino a poco fa il valore era quello
massimo effettivo della giornata e al termine della stessa ripartiva da zero

MrZax75, 2023-09-06:
Questo è il grafico delle ultime ore riguardo a panel max power

commit 6aef13bb (2023-08-17): fix #12