wilds/hass-atonstorage #13: Problem with LogIn

ipisquit, 2023-08-12:
Hi, have same problem with log in, I use the same username and password as I use in the app, and the serial is the alphanumeric serial showed in the app, seems ok but in home assistant all 36 sensors are gone, I have nothing.
the previous version seems ok
thanks

ps121988, 2023-08-13:
> Hi, have same problem with log in, I use the same username and password as I use in the app, and the serial is the alphanumeric serial showed in the app, seems ok but in home assistant all 36 sensors are gone, I have nothing. the previous version seems ok thanks

same problem too

ps121988, 2023-08-17:
> Can you check if there are any errors in the log? I can't reproduce the problem

Unfortunately i rolled back Home Assistant at previuos version.Let's wait for ipisquit , if he won't reply till tomorrow , i'll upgrade again and i'll send you the logs.

ipisquit, 2023-08-17:
Hi, if I check in the log in the left menu, I don't see anything associated with the AtonStorage device.
if i got the wrong log let me know i check

wilds, 2023-08-19:
Left menu: Settings -> System -> Log
Menu laterale: Impostazioni ->Sistema -> Registri

ipisquit, 2023-08-23:
Here the error

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:619 
Integration: Sensore ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22)) 
First occurred: 21 agosto 2023, 15:39:21 (2 occurrences) 
Last logged: 21 agosto 2023, 15:39:21

Error adding entities for domain sensor with platform atonstorage
Error while setting up atonstorage platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 510, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 619, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'serial_number'

wilds, 2023-08-25:
This bug Is fixed in release [v1.0.3](https://github.com/wilds/hass-atonstorage/releases/tag/v1.0.3)

closed 2023-08-25 by wilds

ps121988, 2023-08-26:
Issue isn't solved also after upgrade to  1.0.3 .
In attachement a screenshot after upgrading to latest version of HA
[image]

Here the error by the registry 

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:619
Integration: Sensore (documentation, issues)
First occurred: 22:53:19 (2 occurrences)
Last logged: 22:53:19

Error adding entities for domain sensor with platform atonstorage
Error while setting up atonstorage platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 510, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 619, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'serial_number'

A disposizione !

wilds, 2023-09-05:
Potesti provare a rimuovere completamente, riavviare e reinstallare?
Purtroppo non riesco a replicare il problema