wilds/hass-atonstorage #14: Problem with login ( part 2 )

ps121988, 2023-08-26:
Issue isn't solved also after upgrade to 1.0.3 .
In attachement a screenshot after upgrading to latest version of HA

[image]

Here the error by the registry

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:619
Integration: Sensore (documentation, issues)
First occurred: 22:53:19 (2 occurrences)
Last logged: 22:53:19

Error adding entities for domain sensor with platform atonstorage
Error while setting up atonstorage platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 510, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 619, in _async_add_entity
device = dev_reg.async_get(self.hass).async_get_or_create(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'serial_number'

A disposizione !

wilds, 2023-09-05:
Potesti provare a rimuovere completamente l'integrazione, riavviare e reinstallare?
Purtroppo non riesco a replicare il problema

ps121988, 2023-09-07:
cancellando il tutto e reinstallando ha funzionato.
Grazie

closed 2023-09-07 by ps121988