wilds/hass-atonstorage #22: Daily energy sold incorrect

nspickles, 2023-11-28:
I have configured the energy dashboard in HA using the available sensors but the Daily Energy Sold sensor doesn't seem to be linked up to the energy export value, consequently I am showing no export to grid so the overall numbers are wrong. 

The Power grid input shows that export has occurred today but the daily energy sold value remains at 0.
The daily energy bought value is fine.

nspickles, 2023-12-12:
Any update to this, still an issue in v1.04

ps121988, 2024-02-03:
> I have configured the energy dashboard in HA using the available sensors but the Daily Energy Sold sensor doesn't seem to be linked up to the energy export value, consequently I am showing no export [...]

Same problem too . Is it possible to solve the problem ? 
I'm available to upload any logs.

wilds, 2024-04-08:
#29

closed 2024-04-08 by wilds