wilds/hass-atonstorage #25: Battery Charged/Discharged Energy

nspickles, 2023-12-12:
Prior to the update to v1.04, I had the energy dashboard set to use Battery Charged Energy & Battery Discharged Energy sensors to show how much power had been added or drawn from the battery in kWh.

Since the update to 1.04, neither of these sensors is incrementing even though I am charging batteries of peak and running off battery all day.

Consequently the energy distribution today is now incomplete and I can't see where else to get the battery charge/discharge kWh numbers from.

mauriziosacca, 2023-12-13:
Check in the energy dashboard settings that the entities are set correctly. Some entities should have changed names with the latest update and the dashboard needs to be reconfigured. Are the entities displayed correctly in the integration settings? If not, try deleting and recreating the integration and then check the names of the entities in the dashboards.

PS: at the time of writing there seems to be a problem with AtonStorage, server side. Even the App and website don't work. Wait for it to work normally again before trying on Home Assistant, otherwise you won't understand whether you have solved the problem or not

nspickles, 2023-12-15:
Is the Aton API still broken/down?

The data in the HA integration shows last updated 2 days ago.

There were problems with their servers wed/thu as it affected my PV system but the app and website appears to be all functioning again, I’m just not getting any new data through HA

Cheers,

Neil Pickles
***@***.***<https://www.csy.co.uk/>
T: 0115 911 1677  M: 07801 316711   F: 0115 959 0959
--------------------------------------------------------------------------------------
[email-twitter]  @csyretail<http://www.twitter.com/csyretail>            [email-globe]   www.csy.co.uk<http://www.csy.co.uk/>               [email-mail] ***@***.******@***.***>

mauriziosacca, 2023-12-15:
Yes, it's working again.  Have you tried reloading the integration?  Then do the checks I told you before

nspickles, 2023-12-15:
Thanks for confirming.

I have reloaded the integration and that seems to have fixed the update issue, now showing updated a few seconds ago.

Checked/reselected all the sensors in the energy dashboard grid import/export, battery charge/discharge and PV generation, they all appear to have been correct.

I’ll let it run for a while and see if the problematic values increment or not.

Cheers,

Neil Pickles

nspickles, 2023-12-19:
It has been a few days now since Aton had their issue, the battery charged & discharged energy sensor numbers are not incrementing at all since the update to 1.04. The Daily Sold Energy sensor isn't showing any activity at all even though I exported 4.7kWh today.

nspickles, 2023-12-20:
After deleting everything and starting again the battery charged/discharged energy numbers now appear to be behaving again and incrementing as expected.

It is just the Daily Sold Energy that is still not recording anything as before.

tetionet, 2024-01-05:
I've noted that if I change device name in HA everything will work fine except for battery charged. So now I'm using the device with "username" prefix as currently foreseen in this repository.

I confirm too that Daily Sold Energy is not working (to me it never worked).

closed 2024-05-17 by wilds