wilds/hass-atonstorage #28: Aggiornamento valori non costante

MantovaniMatteo, 2024-03-09:
Salve,
dopo aver aggiornato all'ultima versione di HA:
Core 2024.3.0
Supervisor 2024.02.1
Operating System 12.0
Frontend 20240306.0

**Vedo che i valori non si aggiornano ogni 30 secondo ma dopo circa 10-15minuti.**

Guardando il log compare:

Error doing job: Task exception was never retrieved
Logger: homeassistant
Source: custom_components/atonstorage/**sensor.py:73**
integration: AtonStorage ([documentation](https://github.com/wilds/hass-atonstorage), [issues](https://github.com/wilds/hass-atonstorage/issues))
First occurred: 6:29:06 PM (60 occurrences)
Last logged: 7:06:55 PM

e in fondo a questo log:

File "/config/custom_components/atonstorage/sensor.py", line 581, in native_value
    value = self.entity_description.value_conversion_function(value)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/sensor.py", line 73, in <lambda>
    value_conversion_function=lambda value: min(100, int(value)),
                                                     ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '69.6'

Nel file sensor.py compare questo:

    # BATTERY
    AtonStorageSensorEntityDescription(
        key="soc",
        translation_key="soc",
        name="Battery level",
        native_unit_of_measurement=PERCENTAGE,
        device_class=SensorDeviceClass.BATTERY,
        state_class=SensorStateClass.MEASUREMENT,
        # Limit battery_level to a maximum of 100 and convert it to an integer
        **value_conversion_function=lambda value: min(100, int(value)),**
    ),

Può essere questo errore che non rende possibile l'aggiornamento corretto?
Prima dell'aggiornamento era perfetto, ogni 30" si aggironava!

Volevo capire se il problema è presente solo nella mia integrazione o anche altri stanno avendo lo stesso problema.

Come posso risolvere?

Grazie in anticipo

Matteo

marcoCasamento, 2024-03-09:
Si lo dava anche a me, aton ha iniziato a dare i decimali per la
percentuale di batteria.
Basta mettere un cast a float prima del cast a int, in sensor.py

value_conversion_function=lambda value: min(100, int(float(value)))

Appena sono al PC faccio una PR, intanto se vuoi...

MantovaniMatteo, 2024-03-10:
Ciao Marco, 
grazie mille del suggerimento!

Ho messo:

**value_conversion_function=lambda value: min(100, float(value)),**

perchè inserendo:

value_conversion_function=lambda value: min(100, int(float(value))),

non mi importava più il valore della batteria.

Adesso non ho più l'errore nel Log e aggiorna ogni 30".

Sono troppo contento!

Grazie ancora!

Buona giorrnata

Matteo

closed 2024-04-08 by wilds