wilds/hass-atonstorage #29: Energia Venduta sempre 0

MantovaniMatteo, 2024-03-10:
Buongiorno,
è da circa ottobre/novembre 2023 che ATON non pubblica il valore dell'energia venduta nel dato **eVenduta** (guardando il file get_monitor.php).

Cercando di risolvere in altri modi, Ho provato ad integrare con la  **platform: filter** di home assistant il valore **instant_grid_power_output** associato a **pReteOut** ma non riesco ad avere un valore reale a causa di spike nella potenza, anche usando diversi filtri.

Attulmente ATON fornisce il valore dell'energia venduta nel dato **tot_pReteOut** che è dentro:

_https://www.atonstorage.com/atonTC/get_energy.php_

e non più dentro:

_https://www.atonstorage.com/atonTC/get_monitor.php_

Ho provato a guardare il codice e credo occorrà modificare il file controller.py ma non ho idea!
Non riesco a capire come importare questo valore perchè credo sia necessario inserire poi la data e l'ora.

Qualcuno ha risolto?

Grazie in anticipo e buona giornata

Matteo

ps121988, 2024-03-17:
Ciao Matteo

Anche io sono nelle tue condizioni, possibile che non sia ancora risolto?!

maatx, 2024-03-20:
Idem anche io

Fredxp73, 2024-03-20:
Funzionato bene fiono a ieri ..... dopo aver ricaricato l'integrazione .... oggi mi ritrovo con Energia venduta a 0,
in compenso pero sembra essere ripartito il contatore "Battery Discharged Energy"  (che non ha mai funzionato)   ma non il  " Battery Charged Energy" ....
bhooooooo :-)

marcoCasamento, 2024-03-20:
Ma sarebbe _daily_sold_energy?
Avete errori nei log?

Fredxp73, 2024-03-21:
Ciao Marco,  ti confermo che il sensore che ha smesso di funzionare e' il "_daily_sold_energy" ....... se ti puo essere di aiuto ti fornisco queste informazioni:
Fino a ieri  il "daily_sold_energy"  ha sempre funzionato correttamente
 .... ieri per cercare di far funzionare i 2 sensori " Battery Disch\cahrg Energy" (che non avevano mai funzionato) seguendo alcuni suggerimenti nei post precedenti, ho cancellato e ricaricato il dispositivo (inverter) dall intergrazione ....
si e' tutto ripristinato correttamente,  ma:
1 Energy sold ha smesso di funzionare tutto a 0
2 i  sensori "Battery Disch\cahrg Energy" ora sembrano funzionare correttamente entrambi ....
3 tutti i sensori hanno mantenuto il nome che avevano prima del "ripristino", ma ho perso tutto lo storico della produzione solare, connsumi, etc etc,   
nel file di log ho trovato questo: (riguardo Atonstorage)

TypeError: 'NoneType' object is not subscriptable
2024-03-20 15:55:19.045 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-03-20 15:55:19.665 WARNING (MainThread) [homeassistant.const] ELECTRIC_CURRENT_AMPERE was used from atonstorage, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfElectricCurrent.AMPERE instead, please create a bug report at https://github.com/wilds/hass-atonstorage/issues
2024-03-20 15:55:19.676 WARNING (MainThread) [homeassistant.const] ELECTRIC_POTENTIAL_VOLT was used from atonstorage, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfElectricPotential.VOLT instead, please create a bug report at https://github.com/wilds/hass-atonstorage/issues
2024-03-20 15:55:19.684 WARNING (MainThread) [homeassistant.const] FREQUENCY_HERTZ was used from atonstorage, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfFrequency.HERTZ instead, please create a bug report at https://github.com/wilds/hass-atonstorage/issues

ZeroDivisionError: division by zero

 File "/config/custom_components/atonstorage/sensor.py", line 474, in native_value
    value = self.entity_description.value_calc_function(self.controller)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/sensor.py", line 330, in <lambda>
    value_calc_function=lambda controller: round(100 - ((int(controller.bought_energy) / int(controller.consumed_energy)) *100), 2),
                                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

 File "/config/custom_components/atonstorage/sensor.py", line 474, in native_value
    value = self.entity_description.value_calc_function(self.controller)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/sensor.py", line 330, in <lambda>
    value_calc_function=lambda controller: round(100 - ((int(controller.bought_energy) / int(controller.consumed_energy)) *100), 2),
                                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Fredxp73, 2024-03-21:
eccomi... allora spero di aver fatto correttamente il copia e incolla  :-) comq l'ho inserita cosi:
value_calc_function=lambda controller: (100 if int(controller.consumed_energy) == 0 else round(100 - ((int(controller.bought_energy) / int(controller.consumed_energy)) *100), 2),  
nella linea 330   ma non e' cambiato nulla  .....
nel log ora da solo questo errore !?

024-03-21 13:03:29.910 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON

grazie

marcoCasamento, 2024-03-21:
Beh direi che qualcosa è cambiato, l'errore di divisione per 0 non c'è più,
tuttavia l'avviso che ottieni indica che non riesce a interpretare quello
che riceve dal server.
Se non c'è altro nei log, Potrebbe trattarsi di un un problema di
autenticazione, prova a riavviare home assistant (riavvio completo, non
quello "rapido")

Fredxp73, 2024-03-21:
dopo il riavvio(completo) di HA...   mi da questi errori

TypeError: 'NoneType' object is not subscriptable
2024-03-21 15:38:18.418 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-03-21 15:38:18.880 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.atonstorage.sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1025, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1058, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 991, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1129, in get_code
  File "<frozen importlib._bootstrap_external>", line 1059, in source_to_code
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/atonstorage/sensor.py", line 53
    INVERTER_SENSOR_DESCRIPTIONS = (
                                   ^
SyntaxError: '(' was never closed
2024-03-21 15:38:18.890 ERROR (MainThread) [homeassistant.setup] Unable to prepare setup for platform 'atonstorage.sensor': Platform not found (Exception importing custom_components.atonstorage.sensor).

Fredxp73, 2024-03-21:
> Ok, non vorrei che ci sia qualche errore nel codice che ti ho girato, facciamo che più tardi ti passo l’intero file “sensor.py” testato da me e vediamo

Ciao marco, ti premetto che in programmazione sono a 0 anzi sotto 0 😊 … pur lavorando in nel settore informatico… detto questo …. ho sistemato gli spazi vuoti e le virgole della stringa che mi hai fornito, ora dopo un paio di riavvii, non ho piu nessun errore da parte dell’ integrazione aton , ma l’ energy sold e’ sempre a 0.
Volevo di nuovo ringraziarti il software aton che l’ anno scorso mi hai inviato, non sai quanto mi sta tornando utile.
Sono diversi mesi che ho problemi con l’inverter e quelli di Aton continuano a spostare il problema su un ipotetico ombreggiamento di un pannello…   ma il problema che l’ inverter passa da potenze solari di 3kw a 200w … magari per 1 ora,
e poi all improvviso riparte a 3.5kw …… o viceversa.
Nel frattempo avevo anche installato la scheda modbus via rs485 …. ma con la tabella che ti forniscono loro  ci si fai poco o nulla,
l’unico vantaggio e’ essere svincolati dal cloud e dalle sue latenze, avrei la possibilita’ di prendere il valore dell energia ceduta dalla 485 ..ma dalla loro tabella dei registri ho :

eCeduta High   sta fisso a 2

eCeduta Low  ora segna 29. 994wh  e non mi ritrovo con nessun valore di energia ceduta realmente ….

Ti allego comq il pdf con la tabella dei registri magari un giorno ti puo tornare utile

Grazie a presto

Federico

File "", line 1387, in _gcd_import
File "", line 1360, in _find_and_load
File "", line 1331, in _find_and_load_unlocked
File "", line 935, in _load_unlocked
File "", line 991, in exec_module
File "", line 1129, in get_code
File "", line 1059, in source_to_code
File "", line 488, in _call_with_frames_removed

ps121988, 2024-03-21:
Ciao 
Se trovate l'inghippo, potresti gentilmente rilasciare una nuova release con eventuali istruzioni se c'è qualcosa da editare ?

Grazie

Fredxp73, 2024-03-21:
ciao marco ti ho inviato un messaggio su Gmail .... comq dopo un piccolo ritocco alla riga, sono spariti gli errori ..... ma l 'energy sold sempre a 0 .....

marcoCasamento, 2024-03-21:
Ciao Federico, ok, bene, quindi se l'errore è scomparso è già un passo avanti. Non ce ne sono altri, giusto ? 
Ma a te cosa arriva come eVenduta nella risposta di get_monitor.php ? dagli un occhiata. 
La puoi vedere con gli strumenti di sviluppo del browser (F12) del PC mentre sei collegato al sito di ATON. 
[image]

FrancescoFdd5, 2024-03-21:
Ciao,

Io ci ho guardato ora e viene cosi

[image]

FrancescoFdd5, 2024-03-21:
Aggiungo che se guardo il get_energy viene fuori correttamente 

[image]

Fredxp73, 2024-03-21:
A me cosi .... Venduta 0

[image]

marcoCasamento, 2024-03-21:
Ok allora è chiaro che per il problema di @Fredxp73 non c'è nulla da fare nell' integrazione, mi spiace. Potrebbe essere utile invece una PR per correggere il problema.
@Fredxp73 vuoi pensarci tu? Altrimenti no problema, domani/nel weekend la faccio io.

Fredxp73, 2024-03-21:
Grazie marco se puoi fare la richiesta te e' meglio , io non saprei cosa richiedere :-) .....  la cosa che non mi spiego (da profano) ,,,, fino a ieri questo sold energy che ha funzionato senza problemi da sempre  .....  dopo un semplice ripristino .. ha smesso di funzionare .... e i due sensori batt.charg\disch energy che non hanno mai funzionato all improvviso sono rinati  ..... mah ......i famosi misteri dell informatica heheheheheheh

marcoCasamento, 2024-03-21:
Nessun problema, faccio io.

Boh, ma è chiaro che dipende da Aton, il ripristino dell' integrazione non c'entra nulla.
Secondo me, pura congettura, ci sono server con software diversi, magari stanno aggiornando qualcosa: se si becca un server si ottiene il campo valorizzato, se si becca l'altro no.
Io attenderei di capire se il problema inizia man mano a interessare tutti, segno che l'ipotetico aggiornamento è destinato a restare.

ps121988, 2024-03-22:
Quindi se ho capito bene il problema è che Aton ha cambiato il modo di fornire quel dato e l'autore dell'integrazione deve modificare lo script . corretto ?
sembra che l'autore non sia così responsivo .
Qualcuno ha il suo contatto ?

maatx, 2024-04-02:
nessuna buona nuova su questo problema?

wilds, 2024-04-05:
Proverò a lavorarci questo fine settimana

ps121988, 2024-04-08:
è stato risolto il problema ? non mi sembra di vedere nuove release

closed 2024-04-08 by wilds