wilds/hass-atonstorage #30: Valori differenti

FrancescoFdd5, 2024-03-21:
Ciao a me succede una cosa molto strana. Sto migrando integrazione da quella di LucaPatarca alla tua, ma i valori giornalieri mi vengono riportati in modo corretto all'interno dell'entità. Ma se messi nella plancia energia oppure in un plugin custom come quello nella foto diventano differenti. 

[image]
[image]

Io proprio non capisco quale sia il problema

Fredxp73, 2024-03-21:
Guarda io ho lo stesso problema,  usanndo "Energy Flow control Card Plus" .... dal grafico sono sballati i dati, pero non e' un problema di dati presi dall integrazione, perche io sono conneso diretto con l' inverter  via modbus ... e pure i dati presi diretti da li se inserisco quei sensori me li riporta sballati, credo sia un bug di flow control plus....

FrancescoFdd5, 2024-03-21:
> Guarda io ho lo stesso problema, usanndo "Energy Flow control Card Plus" .... dal grafico sono sballati i dati, pero non e' un problema di dati presi dall integrazione, perche io sono conneso [...]

Intanto grazie,
Volevo farti una domanda che non c’entra…
Come ti sei interfacciato in modbus?

Tornando in topic, la cosa che mi fa stranissimo che è avevo gli stessi valori sballati anche bella plancia energia e con l’altra integrazione va tutto bene

maatx, 2024-03-25:
> Guarda io ho lo stesso problema, usanndo "Energy Flow control Card Plus" .... dal grafico sono sballati i dati, pero non e' un problema di dati presi dall integrazione, perche io sono conneso [...]

Interessa anche a me l'interfacciamento via modbus, si può evitare il cloud?
Grazie

Fredxp73, 2024-03-25:
Ciao ,  devi acquistare la skeda dati da Aton, che pero' non la vende a privati, devi farla ordinare dal tuo elettricista\installatore di fiducia... montarla  e' una fesseria ... va inserita in uno slot....  non so quanto consigliarla pero' .... 
con la tabella registri che ti mandano loro,  ci sono meno sensori di quello che ottieni dall integrazione, 
alcuni valori ottenuti ,  tipo energia ceduta \ acquistata  sono indecifrabili (almeno x me) cioe' non trovo nessuna corrispondenza con il valore del cloud o tramite la pinza amperometrica della shelly

wilds, 2024-04-08:
chiudo, non è un problema dell'integrazione

closed 2024-04-08 by wilds