wilds/hass-atonstorage #32: error installation

danielevsllsv, 2024-04-18:
[%key:common::config_flow::error::unknown%]

set the user,pass and serial this error appears

Paolo7297, 2024-05-16:
I have the same error.
The log says:
```python
Traceback (most recent call last):
  File "/config/custom_components/atonstorage/config_flow.py", line 70, in async_step_user
    await self.async_set_unique_id(slugify(controller.serial_number))
                                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/controller.py", line 214, in serial_number
    return self.data["serialNumber"]
           ~~~~~~~~~^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

Ejl974, 2024-09-16:
Same error for me.

Paolo7297, 2024-09-20:
I solved by going to aton website and login. If you have never changed credentials, it will ask for new password (to change default one), then it worked.

marcels1990, 2024-10-12:
> I solved by going to aton website and login. If you have never changed credentials, it will ask for new password (to change default one), then it worked.

Confirmed, it also worked for me. Thank you very much

closed 2024-12-28 by wilds