wilds/hass-atonstorage #33: Battery percentage

SimoFiuz, 2024-05-10:
Hello all!

Not directly related to the integration itself; I am building a Grafana visualization for my Aton Green Storage.
Even if I am not using hass, your code has been the basis for almost everything, so thank you for sharing. VERY appreciated!

One thing I can't understand is where do you get the Battery percentage (or battery charge). 
In the monitor endpoint there is no such value, or I did not find it.
I would guess at this point it's just calculated from another metric?

Grazie!

marcoCasamento, 2024-05-10:
It's the SOC field in the json answer of get_monitor.php

SimoFiuz, 2024-05-10:
Thanks Marco. 
I was in fact thinking soc means Status of Charge.

However I have been monitoring the value since last night and mine seems broken. It fluctuates between 49 and 51 while the battery has gone from 80% to 30% this morning.

marcoCasamento, 2024-05-10:
That value is what the aton storage web UI uses to display battery status. Do you find that the value returned from the integration and the battery display in the web UI and app are consistent?

SimoFiuz, 2024-05-11:
Ciao Marco,

I started monitoring the "soc" property as soon as I read your message.
There is no percentage indicator in the website, only a battery icon that fills up.

[image]

I am comparing the API property with the percent returned by the Android application.
What I see now:

[image]

Values are off, as you can see above the API reports 53% while the Android application 40%.
But the behavior of the two values is similar, they quickly ramp up when the battery is empty to taper off after 90% for what I suppose is drip charge.

At this point I suppose there are different calculations behind.
I am happy with using the API one, so thank you for your help. I could not find it because it seemed so different than the one I was seeing that I though it must have been something else.

wilds, 2024-05-15:
From dashboard src file https://www.atonstorage.com/atonTC/pages/scripts/index3Aton.js?v=8 :

```
				if (litio==0){
					if (offGrid==0){
						if (isSton(sn))
						{
							if (soc<=35){
								socIcon="ston0";
							}else if (soc<=50){
								socIcon="ston20";
							}
							else if (soc<=65){
								socIcon="ston40";
							}
							else if (soc<=80){
								socIcon="ston60";
							}
							else if (soc<=95){
								socIcon="ston80";
							}
							else {
								socIcon="ston100";
							}
						}
						else{
							if (soc<=35){
								socIcon="bat010";
							}else if (soc<=42.5){
								socIcon="bat020";
							}
							else if (soc<=50){
								socIcon="bat030";
							}
							else if (soc<=57.5){
								socIcon="bat040";
							}
							else if (soc<=65){
								socIcon="bat050";
							}
							else if (soc<=72.5){
								socIcon="bat060";
							}
							else if (soc<=80){
								socIcon="bat070";
							}
							else if (soc<=87.5){
								socIcon="bat080";
							}
							else if (soc<=95){
								socIcon="bat090";
							}
							else {
								socIcon="bat100";
							}
						}
					}else{
						if (soc<=50){
							socIcon="bat010";
						}else if (soc<=55.626){
							socIcon="bat020";
						}
						else if (soc<=61.25){
							socIcon="bat030";
						}
						else if (soc<=66.875){
							socIcon="bat040";
						}
						else if (soc<=72.5){
							socIcon="bat050";
						}
						else if (soc<=78.128){
							socIcon="bat060";
						}
						else if (soc<=83.75){
							socIcon="bat070";
						}
						else if (soc<=89.375){
							socIcon="bat080";
						}
						else if (soc<=95){
							socIcon="bat090";
						}
						else {
							socIcon="bat100";
						}
					}
				} else {
					if (isSton(sn))
					{
						if (soc<=20){
							socIcon="ston0";
						}else if (soc<=38){
							socIcon="ston20";
						}
						else if (soc<=56){
							socIcon="ston40";
						}
						else if (soc<=74){
							socIcon="ston60";
						}
						else if (soc<=92){
							socIcon="ston80";
						}
						else {
							socIcon="ston100";
						}
					}
					else
					{
						**if (soc<=20){
							socIcon="bat010";
						}else if (soc<=29){
							socIcon="bat020";
						}
						else if (soc<=38){
							socIcon="bat030";
						}
						else if (soc<=47){
							socIcon="bat040";
						}
						else if (soc<=56){
							socIcon="bat050";
						}
						else if (soc<=65){
							socIcon="bat060";
						}
						else if (soc<=74){
							socIcon="bat070";
						}
						else if (soc<=83){
							socIcon="bat080";
						}
						else if (soc<=92){
							socIcon="bat090";
						}
						else {
							socIcon="bat100";
						}**
					}
					if (isTrifase(sn)){
						if (soc<=10){
							socIcon="bat010";
						}else if (soc<=19){
							socIcon="bat020";
						}
						else if (soc<=28){
							socIcon="bat030";
						}
						else if (soc<=37){
							socIcon="bat040";
						}
						else if (soc<=46){
							socIcon="bat050";
						}
						else if (soc<=55){
							socIcon="bat060";
						}
						else if (soc<=64){
							socIcon="bat070";
						}
						else if (soc<=73){
							socIcon="bat080";
						}
						else if (soc<=82){
							socIcon="bat090";
						}
						else {
							socIcon="bat100";
						}
					}
				}
```
it should be sufficient to map the range 0-100 onto the range deducible from the code.

SimoFiuz, 2024-05-15:
Thank you very much @wilds . Very appreciated.