wilds/hass-atonstorage #34: Variabili non disponibili dopo aggiornamento HA

FrancescoFdd5, 2024-07-03:
Da circa le 20:40 di oggi, 3 luglio 2024, tutte le variabili sono diventate "non disponibili". Ho provato a ricaricare l'integrazione e a riavviare Home Assistant, ma il problema persiste.

Dal log viewer vedo il seguente log:

`AtonStorageIntegrationSensor(
File "/config/custom_components/atonstorage/sensor.py", line 529, in __init__
  super().__init__()
TypeError: IntegrationSensor.__init__() missing 1 required keyword-only argument: 'max_sub_interval'
2024-07-03 23:42:31.456 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON
`

**Passaggi per riprodurre il problema:**

Aggiornare Home Assistant all'ultima versione disponibile.
Controllare lo stato delle variabili dell'integrazione AtonStorage.
Provare a ricaricare l'integrazione e riavviare Home Assistant.

**Comportamento atteso:**
Le variabili dovrebbero essere disponibili e funzionanti come prima dell'aggiornamento.

**Comportamento osservato:**
Tutte le variabili risultano "non disponibili" e i log riportano gli errori sopra indicati.

**Domanda:**
È un problema dell'integrazione a seguito dell'ultimo aggiornamento di Home Assistant rilasciato stasera?

Grazie per l'aiuto.

Fredxp73, 2024-07-04:
Salve, anche a me fa la stessa cosa ......

maatx, 2024-07-04:
urca, allora aspetto ad aggiornare alla 2024.7

pietropola, 2024-07-05:
Confermo, dopo aver aggiornato stesso problema.
Son tornato alla versione precedente per il momento.

pierpaolo1972, 2024-07-05:
Stesso problema

maatx, 2024-07-06:
Ci sono novità?
Oggi è uscita la 2024.7.1

FrancescoFdd5, 2024-07-06:
Anche con la 2024.7.1 stesso problema

maatx, 2024-07-06:
Ci sono novità.
Con una semplice modifica a sensor.py si sitema e torna a funzionare

FrancescoFdd5, 2024-07-06:
> Ci sono novità. Con una semplice modifica a sensor.py si sistema e torna a funzionare

ciao riesci a dire che modifica? E nel caso fare un pull request

maatx, 2024-07-06:
non è farina del mio sacco. Mi ha aiutato mio figlio.
[image]

FrancescoFdd5, 2024-07-06:
> non è farina del mio sacco. Mi ha aiutato mio figlio. [...]

Confermo ora funziona,

Basta aggiungere 

max_sub_interval=None

Creo una pull request, ovviamente ringraziando te e tuo figlio :)

FrancescoFdd5, 2024-07-06:
#35

maatx, 2024-07-06:
è lui:

https://github.com/massimopavoni

pierpaolo1972, 2024-07-06:
Intanto grazie, potete darmi più dettagli di dove trovo la riga da aggiungere o moficare?

maatx, 2024-07-06:
> Intanto grazie, potete darmi più dettagli di dove trovo la riga da aggiungere o moficare?

guarda l'immagine che ho allegato

pierpaolo1972, 2024-07-06:
si ok, ma dove la apro quella schermata?

maatx, 2024-07-06:
quella è del mio programma di editing  ma puoi usare tranquillamente l'editor di Home Assistant

Fredxp73, 2024-07-06:
Grazie mille, 
anche a me  ora funziona tutto come prima......

pierpaolo1972, 2024-07-06:
C'ho messo un po' a trovare il sensor.py ma confermo che ora dopo la modifica funziona. Grazie mille

FrancescoFdd5, 2024-07-07:
Aggiungo anche la trascrizione per utenti non italiani
I will also add the transcription for non-Italian users.

A user reported an issue with the AtonStorage integration in Home Assistant after a recent update, causing all variables to become "unavailable." The log indicates a TypeError related to a missing argument. Several users confirmed the same issue. A solution involved modifying `sensor.py` by adding `max_sub_interval=None`. This fix resolved the problem, and a pull request #35  was created to incorporate the change into the main codebase.

You can see the line to add in the pictures above.

shadow107, 2024-07-09:
Grazie , anche io ho risolto il problema.
Volevo chiedere se fosse possibile visualizzare quanta energia in kWh è stata accumulata nella batteria e quanta ne è stata consumata giornalmente, al momento credo che faccia la somma in entrambi i casi da quando è attivo l’impianto . Visto che mi segna come carica oggi 1.2 Mwh. Grazie

maatx, 2024-07-09:
>  Volevo chiedere se fosse possibile visualizzare quanta energia in kWh 

Penso basti moltiplicare _Battery charged current_   per _Battery voltage_  per ottenere il valore in Wh

closed 2024-08-27 by wilds