wilds/hass-atonstorage #36: Home Assistant Core Supervisor update

Majikmonke, 2024-07-07:
Hi

I have noticed that I lose most entities (only have 7 remaining) when I update to the most recent.

This happened with update 7 and the 7.1 but was fine with all the 6.4 versions.

FrancescoFdd5, 2024-07-07:
refer to #34

Majikmonke, 2024-07-07:
I would but my Italian is not as good as my english

FrancescoFdd5, 2024-07-07:
Hi, I have added an explanation in English of the issue and how to resolve it.

Majikmonke, 2024-07-07:
Thank you

Kind regards

Dan Scate

Majikmonke, 2024-07-07:
I still don't understand how to fix it. I don't seem to be able to find that entity to alter. I'm using Home Assistant Green through phone so perhaps that's the issue

FrancescoFdd5, 2024-07-08:
You need to modify the plugin file. I don't know if you can do it from home assistant green. I hope the author approves the update soon

Majikmonke, 2024-07-08:
Hi

I needed the add an add on but all now working.

Many thanks

Kind regards

Dan Scate

wilds, 2024-08-27:
duplicate #34

closed 2024-08-27 by wilds