wilds/hass-atonstorage #37: Warning from HA Log

Lantoit, 2024-10-23:
Entity sensor.xxxxx_daily_sold_energy from integration atonstorage has state class total_increasing, but its state is not strictly increasing. Triggered by state 1.45 (1.48) with last_updated set to 2024-10-23T12:30:41.901851+00:00. Please create a bug report at https://github.com/wilds/hass-atonstorage/issues

previous message:

Logger: custom_components.atonstorage.controller
Source: custom_components/atonstorage/controller.py:132
integration: AtonStorage ([documentation](https://github.com/wilds/hass-atonstorage), [issues](https://github.com/wilds/hass-atonstorage/issues))
First occurred: 14:26:11 (1 occurrences)
Last logged: 14:26:11
REST result could not be parsed

wilds, 2025-01-14:
Ciao, si è riverificato questo problema?
purtroppo credo sia un problema dei dati che arrivano dal server Aton: l'energia è un valore strettamente crescente e non dovrebbe verificarsi la ricezione di un valore inferiore al precedente (ad esclusione del caso di reset che viene gestito da hassio correttamente)

Lantoit, 2025-01-14:
Ciao, 
ad ottobre era comparso quel msg e ho fatto quanto richiesto!
Al momento non si è più ripetuto (o non l'ho trovato nel log).
Grazie per l'interessamento.