wilds/hass-atonstorage #38: Aton Axis.T

barryw6588, 2025-01-08:
Could you add the data for the AxisT Aton EV charger please so I can see the power consumed and when and perhaps if allowed control when it charges my car?

Thanks

Barry

wilds, 2025-01-14:
Ciao, quest'addon è frutto del reverese engineering del client web; purtroppo non ho a disposizione di un account che ha collegato AxisT Aton EV . 
Potresti fornirmi screenshot dell'applicazione web per poter capire se riesco a replicare lo stato. O meglio ancora riesci a fornirmi i dump delle risposte del server?
Grazie

barryw6588, 2025-01-14:
Hi, this addon is the result of reverse engineering of the web client; Unfortunately I don't have an account that has connected AxisT Aton EV.
Could you provide me with screenshots of the web application so I can see if I can replicate the state. Or better yet can you provide me with dumps of the server responses?
Thank you

this is a screen shot from my aton login

[image]
[image]

barryw6588, 2025-01-14:
does this help

[image]

wilds, 2025-01-16:
new release [v1.0.7](https://github.com/wilds/hass-atonstorage/releases/tag/v1.0.7)

barryw6588, 2025-01-16:
Thank you but I’m new and don’t understand what I have to do to update to 1/0.7?

 

Regards

 

Barry

07853 635421

barryw6588, 2025-01-16:
You are a very clever man – HACS has just sent the update.  Thank you

 

 

Regards

 

Barry

07853 635421

closed 2025-01-22 by wilds