wilds/hass-atonstorage #39: [%key:common::config_flow::error::unknown%]

skel84, 2025-01-09:
Hi,

I get this error when trying to add the controller. This shows uo in the logs

```
                                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/controller.py", line 214, in serial_number
    return self.data["serialNumber"]
           ~~~~~~~~~^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

[image]

Thanks

wilds, 2025-01-14:
Ciao, non riesco a replicare il problema. Potresti inviare tutto lo stacktrace dell'errore? Grazie

skel84, 2025-01-14:
eccolo:

```
2025-01-14 12:08:17.118 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON
2025-01-14 12:08:17.261 ERROR (MainThread) [custom_components.atonstorage.controller] 'NoneType' object does not support item assignment
2025-01-14 12:08:17.262 ERROR (MainThread) [custom_components.atonstorage.controller] Unable to fetch data. Response: None
2025-01-14 12:08:17.263 ERROR (MainThread) [custom_components.atonstorage.config_flow] Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/atonstorage/config_flow.py", line 70, in async_step_user
    await self.async_set_unique_id(slugify(controller.serial_number))
                                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/controller.py", line 214, in serial_number
    return self.data["serialNumber"]
           ~~~~~~~~~^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

closed 2025-01-14 by wilds

barryw6588, 2025-01-14:
So have you added a new entity for consumption auto?

Regards

Barry
07853 635421

Closed #39<https://github.com/wilds/hass-atonstorage/issues/39> as completed.