wilds/hass-atonstorage #41: Link to Aton

Majikmonke, 2025-02-14:
Hi guys,

Just wondered what link you have to Aton. Reason for asking is I have uncovered an issue with SoC that Aton are now looking at but they appear very unhappy that I use this in HA etc and are implying I'm doing something wrong by using this. 

In the UK at least there will be a firmware update soon so resolve an issue where the SoC suddenly I creases substantially then a few hours later drops back off again. I can see this has happened a few times over the last few months.

wilds, 2025-02-19:
I have no relationship with Aton. All the work done to make the Home Assistant add-on functional has been the result of reverse engineering and analysis of the messages exchanged between official Aton web application and their backend server.
Additionally, all this work is done as a hobby in my free time.

malcolmhm, 2025-02-19:
Top man!!! Time spent we really appreciate!!

Majikmonke, 2025-02-19:
Agree with Malcolm it's great to have such usability, a shame Aton don't agree and a shame they aren't open and how at when mistakes happen but thanks again for your time in creating this.

closed 2025-02-24 by wilds