wilds/hass-atonstorage #42: Charge battery on demand

ps121988, 2025-02-22:
Hello , Aton Storage app has a functionality called 
Pianificazione Anti blackout but HA integration does not permit it .
Could you provide to create it ?
Thank you so much for your work

wilds, 2025-02-24:
Hi, can you tell me if the functionality is also available on the web dashboard?
https://www.atonstorage.com/atonTC/index.php?demo=1

ps121988, 2025-02-27:
Unfortunately no …

HSephiroth93, 2025-11-01:
> In my knowledge the "Pianificazione Anti blackout " functionality is manually activated from the Aton inverter control panel only locally.

This functionality is remotely activated by Aton and is only accessible through the Aton Storage mobile app (not available on the Web).

[image]
[image]

ps121988, 2025-11-01:
> > In my knowledge the "Pianificazione Anti blackout " functionality is manually activated from the Aton inverter control panel only locally.
> 
> This functionality is remotely activated by Aton and is only accessible through the Aton Storage mobile app (not available on the Web).
> 
> <img alt="Image" width="1344" height="2000" src="https://private-user-images.githubusercontent.com/140343237/508555202-57d8de0f-28d5-4383-8dfa-1283a5969692.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjIwMjc0OTIsIm5iZiI6MTc2MjAyNzE5MiwicGF0aCI6Ii8xNDAzNDMyMzcvNTA4NTU1MjAyLTU3ZDhkZTBmLTI4ZDUtNDM4My04ZGZhLTEyODNhNTk2OTY5Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTEwMVQxOTU5NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZGJjZDA1OTk1ZjI3ZmNhMzVhM2RiMDk4ODg0M2FmYjFkNmM5M2MwOGUyMjM2YzU4Yzk3MzgwZTE3ZDJmMmVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.2vGJVroPB58QucyKx-TWfdegWjVWEffWC7dbv91QXEI"> <img alt="Image" width="1344" height="2000" src="https://private-user-images.githubusercontent.com/140343237/508555203-e0471d2f-16f5-475c-815a-5caadf1e5e8a.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjIwMjc0OTIsIm5iZiI6MTc2MjAyNzE5MiwicGF0aCI6Ii8xNDAzNDMyMzcvNTA4NTU1MjAzLWUwNDcxZDJmLTE2ZjUtNDc1Yy04MTVhLTVjYWFkZjFlNWU4YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTEwMVQxOTU5NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hOTY3MDFhYTcxOGE0NDA1YWJkNzU5MDI1OTczNGY5YTllYjZmM2QyOTEyMTkxYWMxNjA1MDRmMjExNzc0NjVhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.OEA_ufgaeWVPXzjjEYM9wCEsVwU68uVbEv9nWNkt3-g">

I wrote several months ago to customer service asking for implementation on web . No answer .
Comunication with customer sucks …