wilds/hass-atonstorage #45: Some errors in HA registry

Dev-Fabry, 2025-03-01:
Hi
These are probably sensors that are not available on my system or do not work as the code would expect

Source: components/sensor/recorder.py:340

Entity sensor.tatale_ev_charged from integration atonstorage has state class total_increasing, but its state is negative. Triggered by state -1.0 with last_updated set to 2025-03-01T20:39:59.999999+00:00. Please create a bug report at https://github.com/wilds/hass-atonstorage/issues

source: helpers/entity_platform.py:738

atonstorage: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 738, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1318, in async_device_update
    await self.async_update()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 601, in async_update
    await self.coordinator.async_request_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 275, in async_request_refresh
    await self._debounced_refresh.async_call()
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 114, in async_call
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 356, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 479, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 559, in _handle_coordinator_update
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/sensor.py", line 559, in native_value
    value = self.entity_description.value_conversion_function(value)
  File "/config/custom_components/atonstorage/sensor.py", line 454, in <lambda>
    value_conversion_function=lambda value: min(100, float(value)),
                                                     ~~~~~^^^^^^^
TypeError: float() argument must be a string or a real number, not 'NoneType'

Source: helpers/entity_platform.py:928

Error adding entity sensor.tatale_ev_charged_percentage for domain sensor with platform atonstorage
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 928, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1384, in add_to_platform_finish
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/sensor.py", line 559, in native_value
    value = self.entity_description.value_conversion_function(value)
  File "/config/custom_components/atonstorage/sensor.py", line 454, in <lambda>
    value_conversion_function=lambda value: min(100, float(value)),
                                                     ~~~~~^^^^^^^
TypeError: float() argument must be a string or a real number, not 'NoneType'

Source: custom_components/atonstorage/controller.py:132

REST result could not be parsed as JSON

wilds, 2025-03-06:
Can you try v1.0.8-beta ? tnx

Dev-Fabry, 2025-03-06:
Yes sure, downloaded but it doesn't work.

Lost all entities and trying to delete the integration and then put it back the configuration menu doesn't come out.

Unable to load configuration flow: {"message":"Invalid handler specified"}

Dev-Fabry, 2025-03-06:
From registry

> Detected blocking call to import_module with args ('custom_components.atonstorage.config_flow',) inside the event loop by integration 'config' at homeassistant/components/config/config_entries.py, line 189: return await super()._post_impl(request, data) (offender: /usr/src/homeassistant/homeassistant/loader.py, line 1303: return importlib.import_module(f"{self.pkg_path}.{platform_name}")), please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+config%22 For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 712, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 683, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2040, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request resp = await request_handler(request) File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle return await handler(request) File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware response = await handler(request) File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle result = await handler(request, **request.match_info) File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin return await func(self, request, *args, **kwargs) File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper return await method(view, request, data, *args, **kwargs) File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 182, in post return await self._post_impl(request, data) File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 189, in _post_impl return await super()._post_impl(request, data) 

and

Registratore: homeassistant.config_entries
Fonte: config_entries.py:3788
Prima occorrenza: 00:04:45 (3 occorrenze)
Ultimo accesso: 00:06:36

> 
Error occurred loading flow for integration atonstorage: cannot import name 'AVAILABLE_SENSORS' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

Dev-Fabry, 2025-03-06:
Reverted to 1.0.7 and now working again

wilds, 2025-03-06:
Sorry, my bad. Try v1.0.8-beta2.

Feel free to reopen issue if it doesn't work

Dev-Fabry, 2025-03-06:
1.0.8 Beta2 installed

It is working but new (or old) errors on registry

This from log viewer at first try

>     return self.data["status"]
>            ~~~~~~~~~^^^^^^^^^^
> TypeError: 'NoneType' object is not subscriptable
> 2025-03-07 00:30:16.599 ERROR (MainThread) [custom_components.atonstorage] Unable to connect to AtonStorage controller: 'NoneType' object is not subscriptable
> 2025-03-07 00:30:26.590 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON

and this at second try (deleted integration and recreated integration)

> 2025-03-07 00:37:01.589 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON
> 2025-03-07 00:37:03.688 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON

Errors only on integration initialize, after for now no errors

Dev-Fabry, 2025-03-06:
New on log now:

> 
2025-03-07 00:40:10.067 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.tatale_ev_charged from integration atonstorage has state class total_increasing, but its state is negative. Triggered by state -1.0 with last_updated set to 2025-03-06T23:37:01.602796+00:00. Please create a bug report at https://github.com/wilds/hass-atonstorage/issues

Dev-Fabry, 2025-03-07:
Full Log after last installation (only one new warning in bold)

> 2025-03-07 00:30:26.590 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON
> 2025-03-07 00:37:01.589 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON
> 2025-03-07 00:37:03.688 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON
> 2025-03-07 00:40:10.067 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.tatale_ev_charged from integration atonstorage has state class total_increasing, but its state is negative. Triggered by state -1.0 with last_updated set to 2025-03-06T23:37:01.602796+00:00. Please create a bug report at https://github.com/wilds/hass-atonstorage/issues
>** 2025-03-07 01:05:05.098 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON**

Dev-Fabry, 2025-03-08:
Log update of today:

> 2025-03-07 12:11:28.568 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON
> 2025-03-07 12:15:10.425 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.xxxx_ev_charged from integration atonstorage has state class total_increasing, but its state is negative. Triggered by state -1.0 with last_updated set to 2025-03-07T11:09:59.999999+00:00. Please create a bug report at https://github.com/wilds/hass-atonstorage/issues
> 2025-03-07 13:35:10.407 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.xxxx_daily_sold_energy from integration atonstorage has state class total_increasing, but its state is not strictly increasing. Triggered by state 3.92 (3.95) with last_updated set to 2025-03-07T12:30:29.371430+00:00. Please create a bug report at https://github.com/wilds/hass-atonstorage/issu

commit ee662d85 (2025-03-06): fix #45