wilds/hass-atonstorage #46: Binary sensors are only updated on startup

Dev-Fabry, 2025-03-02:
Hi
At least with version 1.0.7 the binary sensors remain fixed and do not change to reflect the real state of the system at that given moment.

But if you restart HA they update and take the correct state (and then keep it fixed).
It's as if those sensors were read only once when connecting to the ATON cloud   

I just monitored:
> binary_sensor.dev-id_solar_to_house
> binary_sensor.dev-id_solar_to_battery
> binary_sensor.dev-id_solar_to_grid
> binary_sensor.dev-id_grid_to_house
> binary_sensor.dev-id_grid_to_battery
> binary_sensor.dev-id_battery_to_house
> binary_sensor.dev-id_batter_to_grid
> 
but I think it's the same for everyone else

wilds, 2025-03-06:
Hi, can you check that the "scan interval" value is correct?

Dev-Fabry, 2025-03-06:
In a first instance I had entered 5 sec because I wanted more detailed graphs

Then I saw that from the Cloud only data with 30 sec detail arrives and I reconfigured the integration to use 30 sec, at least this way it filled up the register less with the reported errors

In any case the other entities (e.g. instant_solar_power) update every 30 sec, the binary sensors remain fixed even if there are changes

Example
I have not restarted for hours and as binary sensors turned on I have solar_to_house and battery_to_house, and all the others turned off, but at this moment there is NO solar production

In fact immediately after the HA restart, solar_to_house went to OFF

Dev-Fabry, 2025-03-08:
I don't know if it was deleting and reinstalling the integration or if it's the switch to version 1.0.8-beta2 instead of 1.0.7 but now the binary indicators seem to update

I'll keep monitoring the situation in the next few days

wilds, 2025-03-11:
Can we consider this problem solved? Can I close the issue?

Dev-Fabry, 2025-03-13:
Hi, Yes it can probably be considered closed.

I have not had time to personally monitor the binary sensors to verify that they switch exactly when they should, but the stored history shows that at least they switch and are not fixed, despite there having been no reboots for several consecutive days

closed 2025-03-14 by wilds