wilds/hass-atonstorage #47: Connect to the AtonStorage controller: [%key:common::config_flow::error::unknown%]

fpalladino, 2025-05-20:
Durante l'aggiunta dell'integrazione, inserendo le credenziali ed il seriale esce sempre lo stesso errore.
Provato con la versione 1.0.7 e 1.0.8 beta

fpalladino, 2025-05-20:
2025-05-20 20:04:31.002 WARNING (MainThread) [custom_components.atonstorage.controller] REST result could not be parsed as JSON
2025-05-20 20:04:31.124 ERROR (MainThread) [custom_components.atonstorage.controller] 'NoneType' object does not support item assignment
2025-05-20 20:04:31.124 ERROR (MainThread) [custom_components.atonstorage.controller] Unable to fetch data. Response: None
2025-05-20 20:04:31.124 ERROR (MainThread) [custom_components.atonstorage.config_flow] Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/atonstorage/config_flow.py", line 70, in async_step_user
    await self.async_set_unique_id(slugify(controller.serial_number))
                                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/atonstorage/controller.py", line 214, in serial_number
    return self.data["serialNumber"]
           ~~~~~~~~~^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable

BekaMaster, 2026-09-25:
Stesso problema anch'io! Ho gia impostato la mail e sto utilizzando le stesse credenziali che uso su app e sito, ma l'integrazione mi da sempre questo stesso errore

BekaMaster, 2026-09-25:
Ho scoperto cosa non andava!
Il problema è il seriale!
Sull'inverter c'è una targetta con un seriale di 16 cifre, ma NON è quello corretto da inserire.
Il seriale corretto (recuperato dall'app) è alfanumerico, il mio è formato cosi: X11XX11111X (dove X sono carattteri e 1 numeri)

Una volta inserito quello corretto l'integrazione ha funzionato

Suggerisco di implementare un errore più chiaro e magari aggiungere un hint nel campo dove bisogna inserire il seriale su quale sia il formato