wilds/hass-atonstorage #49: Problema post aggiornamento

ros-mas, 2025-08-07:
Salve, dopo avere aggiornato il core al 2025.8.0 mi sono scomparsi tutti i sensori, provato a reinstallare ma non cambia nulla.

Dev-Fabry, 2025-08-10:
E' un problema di compatibilità tra questa integrazione e la versione 2025.8.x

Devi tornare alla versione 2025.7.4 facendo "core update --version 2025.7.4" dalla CLI di HA

Vedi: [qui](https://github.com/wilds/hass-atonstorage/issues/50#issuecomment-3172620961)

E' stata proposta la soluzione [qui](https://github.com/wilds/hass-atonstorage/pull/51) ma per ora non è ancora stata integrata.

geniodelmale, 2025-08-11:
Ho integrato la pull request #51 qui: https://github.com/geniodelmale/hass-atonstorage potete rimuovere l'originale e usare questo intanto che la pull request viene approvata ufficialmente